Spanni26/pyHPSU #18: query

relft, 2019-04-14:
Hi,

I'm using https://github.com/zanac/pyHPSU with latest versions of june 2018 and rote HPSU 508 , and I have detected some issue, not working daemon (as workaround using a script + sleep) and also receiving some errors like 

2019-04-14 07:52:26,515 - ERROR - RECV:bytearray(b'b\x10\xfa\n\x0c\x00/')
2019-04-14 07:52:26,516 - ERROR - msg not sync, timeout
2019-04-14 07:52:28,520 - ERROR - SEND:[49, 0, 3, 0, 0, 0, 0]
2019-04-14 07:52:28,521 - ERROR - RECV:bytearray(b'2\x10\x0e\x01\x87\x00\x00')
2019-04-14 07:52:28,522 - ERROR - SEND:[49, 0, 3, 0, 0, 0, 0]
2019-04-14 07:52:28,522 - ERROR - RECV:bytearray(b' \nR\x01\x00\x00\x00')
2019-04-14 07:52:28,524 - ERROR - SEND:[49, 0, 3, 0, 0, 0, 0]
2019-04-14 07:52:28,524 - ERROR - RECV:bytearray(b' \nV\x00\x00\x00\x00')
2019-04-14 07:52:28,525 - ERROR - SEND:[49, 0, 3, 0, 0, 0, 0]
2019-04-14 07:52:28,526 - ERROR - RECV:bytearray(b' \nW\x00\x00\x00\x00')
2019-04-14 07:52:28,527 - ERROR - SEND:[49, 0, 3, 0, 0, 0, 0]

My question is related with your version, is the same than zanac. can I use without compatibility issues? what are the differences?

Thanks

relft, 2019-04-14:
With you versions I can use to monitor and also attach a rocon u1 without issues or incompatibilities?

Thanks

Spanni26, 2019-04-15:
spanni/pyHPSU is based on the code of zanac/pyHPSU, but is under active development.
You can connect your can interface parallel to the Rocon U1

relft, 2019-05-11:
Thanks, Connected rocon 1 and keeping connected with pyhpsu and all is working!!!
 Great!

Now I'm using latest Zanac version, and I have chcked that this version Spanni26 is newer and with more features, I can use same  versions with smae commands used in Zanac or  I need to make some changes?

I'm searching how to obtain ambien temperatures from rocon u1 but I can't fins where is stored in Zanac or spanni version, any suggestion how to get insider temperature from rocon u1?

Thanks

Spanni26, 2019-05-12:
As said: this repo hier (spanni26) is based on zanacs pyHPSU. But he can't develop his version anymore cause a lack of time. So we forked his version and developed it further.
So Zanacs version is dead (sorry zanac). But all of his work is here in the spanni26 fork.

What temperature do you wan't to read? The temperature in the house? Do you have a sensor installed for that?

relft, 2019-05-12:
Hi,

Right, in this case I'll try to migrate to your version, any suggestion or
only replace with same actual config?
I want to read in house temperature from rocon u1, I have this temperature
on rocon u1 lcd display but I don't know where to obtain this with pyhpsu
and receive in emoncms... I Also have installed rocon ot 1 (external
temperature) and this is correctly displayed from pyhpsu

Thanks

Missatge de Spanni26 <notifications@github.com> del dia dg., 12 de maig
2019 a les 9:37:

Spanni26, 2019-05-12:
Hello,

I've never seen a rocon U1. Is the temperature always shown on the display or dou you have to select it?
We have to find it in a can dump.

relft, 2019-05-12:
Hi,

Is is always shown  on the display with external rocon ot1, both
temperatures are showed always
If you need I can help to find this in our environment, if you give me
detail I can try to help

Thanks

Missatge de Spanni26 <notifications@github.com> del dia dg., 12 de maig
2019 a les 10:31:

Spanni26, 2019-05-13:
Is the temperature also shown on the HPSU display?

relft, 2019-05-13:
Is shown on  rocon u1 lcd display, always on main menu with rocon ot1
external temperatures, both temp on main lcd display

Thanks

Spanni26, 2019-05-14:
Is the room temperature shown on the Display of the HPSU or can it be displayed there? Otherwise its hard to find the can code

svde, 2019-05-14:
Perhaps it's:
        {"name": "cRAUMISTTEMP",
            "infoNumber": {"byteHigh": "0x00", "byteLow": "0x11"}, "type": "float", "transferThreshold": "0.5", "factor": "10"},

Spanni26, 2019-05-14:
Possibly :) But whats the can code?

svde, 2019-05-14:
$ /home/stefan/can_progs/can_scan can0 680 180.0011
elster-kromschroeder can-bus address scanner and test utility
copyright (c) 2014 JÃ¼rg MÃ¼ller, CH-5524

value: 8000  (RAUMISTTEMP  not available)

Which makes sense, as I don't have the rocon u1 device.

Spanni26, 2019-05-14:
If someone could lend me  U1 I can test it...

gobartos, 2019-05-14:
I have 2 of them in use :)

Spanni26, 2019-05-14:
I need it here :)

svde, 2019-05-14:
Or... you need somebody to run some commands on the canbus connected to a system which has a u1

Spanni26, 2019-05-14:
Its a timing problem....you have to run a candump right at that moment you call the room temperature directly at the HPSU :)

svde, 2019-05-14:
Or just ask for the value using can_scan can0 680 180.0011

Spanni26, 2019-05-14:
can_scan never worked for me. Can you du a "candump can0" and call your can_scan?

svde, 2019-05-14:
can0  69D   [7]  C1 0E FA 01 99 00 00
  can0  680   [7]  31 00 FA 00 11 00 00
  can0  180   [7]  D2 00 FA 00 11 80 00
  can0  69D   [7]  C1 0F FA 01 99 00 00

relft, 2019-05-14:
Hi,

I have a u1, if anybody give details about required information and how to
obtain I can try to get
Thanks

Missatge de svde <notifications@github.com> del dia dt., 14 de maig 2019 a
les 13:34:

Spanni26, 2019-05-15:
Ok. I've added a command "t_room". 
To test it you have to pull the repo and call install.sh
If someone with an rocon U1 could test it please :)

relft, 2019-05-15:
Hi,

It seems that is not working correctly comparing with ot1 external temp, an
example

root@emonpi:/local/pyHPSU# pyHPSU.py -c t_outdoor_ot1
[{'name': 't_outdoor_ot1', 'resp': '16.80', 'timestamp':
1557948828.651261}]  --> Correct
root@emonpi:/local/pyHPSU# pyHPSU.py -c t_room
[{'name': 't_room', 'resp': '-3276.80', 'timestamp': 1557948841.112846}]
--> incorrect value, must be 24.2

If you need I can help if you give details about what do you need

Thanks!

Missatge de Spanni26 <notifications@github.com> del dia dc., 15 de maig
2019 a les 7:50:

Spanni26, 2019-05-16:
I still need the same: The Can-Code and the information, if the room-temp is displayed directly at the HPSU. 
We only talk directly to the controller from the HPSU. If the Controller doesn't know the room temp...we can't ask him about :)

Spanni26, 2019-05-16:
@relft : please do the following:
on you unix machine do:
Make sure you installed the program "candump" (on debian it's contained in "can-utils")
"timout 300 candump can0 > /tmp/candump_relft" and post the file /tmp/candump_relft
Please send us also the temperature showed at your U1

relft, 2019-05-16:
Hi,
Attached file

Missatge de Spanni26 <notifications@github.com> del dia dj., 16 de maig
2019 a les 14:27:

hackepeterOli, 2019-05-16:
Is the room temperature writable so I can write it with my own thermostat?

relft, 2019-05-16:
Do you can read?

Missatge de hackepeterOli <notifications@github.com> del dia dj., 16 de
maig 2019 a les 17:55:

Spanni26, 2019-05-16:
Hmmm, where is that file @relft ?
Please send it to dspannbauer@web.de

Spanni26, 2019-05-17:
Ok, got the Log.
@relft  How many U1 do you have? I have 2 can IDs found in the log which I can't find in my home installation...6A8 and 400

Spanni26, 2019-05-17:
I changed the command for t_room to query the 400....please test it. But write down the major configuration values from your HPSU...just to be save :)

relft, 2019-05-17:
Only one u1

relft, 2019-05-17:
Hi

Sorry, but i don't understand this
But write down the major configuration values from your HPSU...just to be
save :)

Can you give details what you need? Test or not to test?

Spanni26, 2019-05-17:
You can test it....theres always a risk to reset the HPSU when testing values....so take a paper and a pencil :)

relft, 2019-05-17:
Also in this case, when you try to read and bot writing ?

Spanni26, 2019-05-17:
Hmm, no. When you try to write some unknown value. But In your case there is no problem cause you are just reading....

relft, 2019-05-17:
Hi,

Great job seems that is working!

#  pyHPSU.py -c t_outdoor_ot1
[{'name': 't_outdoor_ot1', 'resp': '15.60', 'timestamp': 1558112181.062608}]
# pyHPSU.py -c t_room
[{'resp': '23.10', 'name': 't_room', 'timestamp': 1558112186.986957}]

Now I'll try to migrate my zanac environment to spanni26 versions, I can
keep all historical data, config and replace pyhpsu zanac binaries by
spanni26, wihtout losing any data (and historical data) or configuration,
or it is best to make a new fresh install environment?
If it is possible I like to keeps all previous data, if this keeps same
source and vars for data that is obtained from hpsu

Thanks

Missatge de Spanni26 <notifications@github.com> del dia dv., 17 de maig
2019 a les 11:06:

relft, 2019-05-17:
Hi

Trying to migrate to spanni26 from zanac I have problems to configure emconcms
with zanac using:
 cat emoncms.ini
[config]
apikey=key.....
emoncms_url=http://ip/emoncms

[node]
HPSU508=t_hs,t_hc,flow_rate,t_return,mode,t_ext,t_hc_set,bpv,posmix,t_dhw,t_dhw_set,t_v1,t_dhw1,t_vbh,t_r1,tliq2,tdhw2,ehs,qboh,qchhp,qch,qwp,qdhw,pump,ta2,t_outdoor_ot1,t_dhw_setpoint1,hyst_hp

trying with spanni26 in pyhpsu.conf added:
[EMONCMS]
URL = http://ip/emoncms
APIKEY = key...
HPSU508 = t_hs

and I receive this error: 
# pyHPSU.py -d PYCAN  -c t_hs  -o EMONCMS
warning - command t_hs defined as commandline option but not in emoncms

Any suggestion about where is the error, I have ried adding node_HPSU508 =.. and then error is 
# pyHPSU.py -d PYCAN  -c t_hs  -o EMONCMS
warning - command t_hs defined as commandline option but not in emoncms
Traceback (most recent call last):
  File "/usr/bin/pyHPSU.py", line 339, in <module>
    main(sys.argv[1:])
  File "/usr/bin/pyHPSU.py", line 276, in main
    read_can(driver, logger, port, cmd, lg_code,verbose,output_type)
  File "/usr/bin/pyHPSU.py", line 333, in read_can
    hpsu_plugin.pushValues(vars=arrResponse)
  File "/usr/lib/python3/dist-packages/HPSU/plugins/emoncms.py", line 91, in pushValues
    for r2 in self.listNodes[node]:
TypeError: 'NoneType' object is not iterable

Searching and example of configuration like this https://github.com/zanac/pyHPSU/blob/f056924ad16dd4bdc4900ecbc7db20b7f04d896d/etc/pyHPSU/EMONCMS.ini but I can't find

Any suggestion?
Thanks

relft, 2019-05-18:
with zanac versions command used is 
 sudo /usr/share/pyHPSU/bin/pyHPSU.py -d PYCAN  -c t_hs -c t_hc -c flow_rate -c t_return -c mode -c t_ext -c t_hc_set -c bpv -c posmix -c t_dhw -c t_dhw_set -c t_v1 -c t_dhw1 -c t_vbh -c t_r1 -c tliq2 -c tdhw2 -c ehs -c qboh -c qchhp -c qch -c qwp -c qdhw -c pump -c ta2 -c t_outdoor_ot1 -c t_dhw_setpoint1 -c hyst_hp -o CLOUD -u EMONCMS -g /var/log/pyHPSU_pican2.log -v 1
and this works, 

but with spanni26 versions is not working, seems that -u (upload) has been removed and I can't find a clear information how to configure data to emoncms, anybody knows if this is working or how can I solve the issue?

Thanks

pdcemulator, 2019-05-18:
Do you have something like this in your pyhpsu.conf?
`[EMONCMS]
URL = https://emoncms.org
node_1 = t_hs, t_hs_setpoint
node_2 = qoh....`

relft, 2019-05-18:
Right, tried multiple configurations with this:

[EMONCMS]
URL = http://ip/emoncms
#APIKEY = key
node_HPSU508 = t_hs
##node_1 = t_hs

tried comment node_HPSU508, and enabled node_1 without success...  also tried with and without apikey ... I have chcked code and comapring with zanac -u upload was removed and with zanac was needed

Spanni26, 2019-05-18:
Try pyHPSU.py -?

⁣Gesendet mit BlueMail ​

Spanni26, 2019-05-21:
News here?

relft, 2019-05-21:
No sucess... I ll try to make a fresh install with latest version emonc 10
and test, but now my availability is restricted...
Any example or any user is working with emoncms and spanni26 version? Any
configuration example?

Thanks

Spanni26, 2019-05-21:
How do you call the pyHPSU from the Spanni repo?

⁣Gesendet mit BlueMail ​

relft, 2019-05-21:
Hi,

using an script as a cron job every 15 seconds with this:
sudo /usr/share/pyHPSU/bin/pyHPSU.py -d PYCAN  -c t_hs -c t_hc -c flow_rate
-c t_return -c mode -c t_ext -c t_hc_set -c bpv -c posmix -c t_dhw -c
t_dhw_set -c t_v1 -c t_dhw1 -c t_vbh -c t_r1 -c tliq2 -c tdhw2 -c ehs -c
qboh -c qchhp -c qch -c qwp -c qdhw -c pump -c ta2 -c t_outdoor_ot1 -c
t_dhw_setpoint1 -c hyst_hp -c t_room1_setpoint -c t_room2_setpoint -c
t_room3_setpoint -c t_room -c t_hs_set -c water_pressure -c t_hc -c
t_hc_set -c status_pump -c runtime_comp -c runtime_pump -c qsc -c mode_01
-c tvbh2 -c tr2 -c quiet -c ext -c rt -c v1 -c heat_slope -c t_flow_cooling
-o EMONCMS -g /var/log/pyHPSU_pican2.log -v 1

same command with zanac is:
sudo /usr/share/pyHPSU/bin/pyHPSU.py -d PYCAN  -c t_hs -c t_hc -c flow_rate
-c t_return -c mode -c t_ext -c t_hc_set -c bpv -c posmix -c t_dhw -c
t_dhw_set -c t_v1 -c t_dhw1 -c t_vbh -c t_r1 -c tliq2 -c tdhw2 -c ehs -c
qboh -c qchhp -c qch -c qwp -c qdhw -c pump -c ta2 -c t_outdoor_ot1 -c
t_dhw_setpoint1 -c hyst_hp -c t_room1_setpoint -c t_room2_setpoint -c
t_room3_setpoint -c t_room -c t_hs_set -c water_pressure -c t_hc -c
t_hc_set -c status_pump -c runtime_comp -c runtime_pump -c qsc -c mode_01
-c tvbh2 -c tr2 -c quiet -c ext -c rt -c v1 -c heat_slope -c t_flow_cooling
-o CLOUD -u  EMONCMS -g /var/log/pyHPSU_pican2.log -v 1

I have checked and seems taht cloud options was removed an emoncms also
removed to be usable... in my tests...

Missatge de Spanni26 <notifications@github.com> del dia dt., 21 de maig
2019 a les 21:54:

Spanni26, 2019-05-22:
can you show us your pyhpsu.conf?
I removed the cloud option to make it more clear that there is a plugin for each method of sharing the data.

Spanni26, 2019-05-22:
Will try EMONCMS tomorrow...

relft, 2019-05-26:
Hi,

Now I ahve upgraded emoncms from 9.9.5 to 10.0.2, and next action will be
try if now works... I'll try to report
One question, I suspect that can be a conflict with having both systems
installed on same raspberry (different paths) zanac and spanni26, do you
know if can exista file or files that can generate some conflict?

Thanks

Missatge de Spanni26 <notifications@github.com> del dia dc., 22 de maig
2019 a les 18:18:

Spanni26, 2019-05-27:
Simply delete the directory with the zanac_hpsu to be save. But it shouldn't conflict with the spanni_pyHPSU

Chrischi-, 2019-06-05:
Hi, 

i recently getting more and more "communication error" displayed on the control panel unit and i have to set the whole system back to "config: active". I only read values via CAN and don't write anything. 

I think that "pyHPSU/CAN BUS" uses the same CAN-Endpoint as the control panel. 
Can anyone say something about this? That would also makes sense if you look which answer the original Rocon gets after a read query.

greets

Spanni26, 2019-06-05:
config:active? Where can you set this?
To check if another endpoint uses the sam ID simply shut down the pyHPSU and start a candump. The ID should appear there. 
In the command only the destination of the telegram is set, who you asking about a value.  For sending the ID set in the "ID" tag from the specific command is used. 

I get Comm error when there are a lot of messages on the bus, so if I'm querying a lot of values at once.

Chrischi-, 2019-06-05:
> config:active? Where can you set this?

The Display at the Rotex Unit shows a "Communication Error" and the whole System has to be reconfigured.

svde, 2019-06-05:
Which error is it? E91?

Spanni26, 2019-06-05:
No, which commands causes the error :)

⁣Gesendet mit BlueMail ​

Chrischi-, 2019-06-06:
No, the display does not have an error number. 

As if you were setting up the entire system from factory defaults again.

It seems like after setting `AT SH 310` to get the external temp (`61 00 FA 0C 1F 00 00`).
But the error occurs very infrequently.