Spanni26/pyHPSU #21: value overflow qch - starting again at 0

pdcemulator, 2019-06-10:
Hello,

the value for qch (or qwp a longer time ago) was above 32228 and is starting again at 0. So I suppose an internal numger range was overflowed but I didnt had a value logging active at this time. So I do not know the exact value to add to the current value to get the all time qch value in my system.

Has anybody an data logging active while one of his value get this overflow?

As an IT guy I would expect 32768, but I am not sure, so I am asking here for other experiences.
Thank you

Spanni26, 2019-06-10:
Hmm, looks like the conversation into a real value gets wrong.

Will investigate this tomorrow.

Daniel

pdcemulator, 2019-06-10:
Hi Daniel,
this is not an pyHPSU issue. Rotex device also display also values beginning with 0. So it seems that the internal number storage for this value was full and started again at 0.

Spanni26, 2019-06-12:
Hmmm,

qch is the ammount of the generated heat. Seems to be a firmware problem
of the HPSU.

closed 2020-10-02 by Spanni26