Spanni26/pyHPSU #23: Output Adapter MQTT

adieball, 2019-06-14:
Hi

when using the new (Beta) MQTT Adapter, I get:

`pi@can:~/pyHPSU/plugins $ pyHPSU.py -c water_pressure -o mqtt
Traceback (most recent call last):
  File "/usr/bin/pyHPSU.py", line 335, in <module>
    main(sys.argv[1:])
  File "/usr/bin/pyHPSU.py", line 275, in main
    read_can(driver, logger, port, cmd, lg_code,verbose,output_type)
  File "/usr/bin/pyHPSU.py", line 331, in read_can
    hpsu_plugin = module.export(hpsu=n_hpsu, logger=logger, config_file=conf_file)
  File "/usr/lib/python3/dist-packages/HPSU/plugins/mqtt.py", line 77, in __init__
    if self.username:
AttributeError: 'export' object has no attribute 'username'
pi@can:~/pyHPSU/plugins $ `

adieball, 2019-06-14:
My Fault, I edited the wrong MQTT.py file.

Now I get
`pi@can:~/pyHPSU/plugins $ pyHPSU.py -c water_pressure -o mqtt
Traceback (most recent call last):
  File "/usr/bin/pyHPSU.py", line 335, in <module>
    main(sys.argv[1:])
  File "/usr/bin/pyHPSU.py", line 275, in main
    read_can(driver, logger, port, cmd, lg_code,verbose,output_type)
  File "/usr/bin/pyHPSU.py", line 332, in read_can
    hpsu_plugin.pushValues(vars=arrResponse)
  File "/usr/lib/python3/dist-packages/HPSU/plugins/mqtt.py", line 94, in pushValues
    ret=self.client.publish(self.prefix + "/" + r['name'],payload=r['resp'], qos=self.qos)
  File "/home/pi/.local/lib/python3.5/site-packages/paho/mqtt/client.py", line 1136, in publish
    if qos < 0 or qos > 2:
TypeError: unorderable types: str() < int()
pi@can:~/pyHPSU/plugins $ `

On the MQTT Adapter I can see that rotes connected with the secret but nothing else

Spanni26, 2019-06-19:
Should be fixed.

adieball, 2019-08-21:
Hi. Have you fixed based on the comments I made for the Mqtt.py files

Spanni26, 2019-08-22:
Hmmm....was a while ago :) Don't know.

jockele79, 2020-03-28:
Hello everybody,

I just set up a raspi 3 and have the same problem:

`root@raspberrypi:~# pyHPSU.py -c t_hs -o mqtt
Traceback (most recent call last):
  File "/usr/bin/pyHPSU.py", line 335, in <module>
    main(sys.argv[1:])
  File "/usr/bin/pyHPSU.py", line 275, in main
    read_can(driver, logger, port, cmd, lg_code,verbose,output_type)
  File "/usr/bin/pyHPSU.py", line 330, in read_can
    module = importlib.import_module("HPSU.plugins." + module_name)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/HPSU/plugins/mqtt.py", line 18, in <modul             e>
    import paho.mqtt.publish as publish
ModuleNotFoundError: No module named 'paho'`

Does anyone have a solution for this problem?

Spanni26, 2020-03-28:
Moin, thy python module "phao" is missing, its needed for mqtt.RegardsDaniel

jockele79, 2020-03-28:
`pi@raspberrypi:~ $ sudo pip install paho-mqtt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: paho-mqtt in /usr/local/lib/python2.7/dist-packag             es (1.5.0)
`

Is the problem that's it's installed in python 2.7-folder?
Sorry for the questions, I'm a real linux-layman

pdcemulator, 2020-03-28:
@jockele79 
Please install the Python3 packages as described here: https://github.com/Spanni26/pyHPSU/blob/ced0ddf6be407b23bc9213b43e06d2b880c951e1/README.md

@Spanni26 
Please merge my PR with improved Readme, which already covers this

Spanni26, 2020-03-28:
Sorry, is merged now :)

Spanni26, 2020-03-28:
Yes, thats wrong :)

You need it for python3, not python2.

I would not install it via pip. Use apt or a similar tools (depending on
your distribution)

Regards

Daniel

jockele79, 2020-03-28:
ok, thanks, I tried this and receive the following output:

`root@raspberrypi:/pyHPSU# pyHPSU.py -c t_hs -o mqtt
Traceback (most recent call last):
  File "/usr/bin/pyHPSU.py", line 335, in <module>
    main(sys.argv[1:])
  File "/usr/bin/pyHPSU.py", line 275, in main
    read_can(driver, logger, port, cmd, lg_code,verbose,output_type)
  File "/usr/bin/pyHPSU.py", line 331, in read_can
    hpsu_plugin = module.export(hpsu=n_hpsu, logger=logger, config_file=conf_file)
  File "/usr/lib/python3/dist-packages/HPSU/plugins/mqtt.py", line 77, in __init__
    if self.username:
AttributeError: 'export' object has no attribute 'username'
`

pdcemulator, 2020-03-28:
Please post your config file (/etc/pyHPSU/pyhpsu.conf). It looks like that you do not have configured an username. It should look like this, modify do your setup.

[MQTT]
BROKER = mybroker
PORT = 1883
USERNAME = myuser
PASSWORD = mysecretpass
CLIENTNAME = rotex_hpsu
PREFIX = rotex

jockele79, 2020-03-28:
here my config, I think this should work

# settings for pyHPSUd 
[PYHPSUD]
PYHPSUD_DEVICE=PYCAN
PYHPSUD_LANG=EN
PYHPSUD_PORT=/dev/ttyUSB0

# settings for pyHPSU
[PYHPSU]
PYHPSU_DEVICE = PYCAN
PYHPSU_PORT = /dev/ttyS0
PYHPSU_LANG = DE
OUTPUT_TYPE = MQTT

[MQTT]
# BROKER = 192.168.178.10
# PORT = 1883
# USERNAME = edomi
# PASSWORD = 123456
# CLIENTNAME = rotex_hpsu
# PREFIX = rotex

# interval for each query for a value in seconds
# values not listed here are not queried

[JOBS]

T_HS = 2
T_HS_SET = 25
water_pressure = 4
t_ext = 6
t_dhw = 25
t_dhw_set = 60
t_return = 4
flow_rate = 2
t_hc = 4
t_hc_set = 2

pdcemulator, 2020-03-29:
Remove the # sign before lines in MQTT section - currently they are commented out and were not used

jockele79, 2020-03-29:
the # was already deleted.  
This morning I made a complete reinstallation of the raspi, everything was running correctly. During installation the apt-get install python3-paho-mqtt also was installed successfully. 
I can read the values manually, also the command -a is running, but after configuring MQTT within the config-file I get the following output:

`pi@raspberrypi:/pyHPSU $ pyHPSU.py -c t_hs -d PYCAN -o MQTT
Traceback (most recent call last):
  File "/usr/bin/pyHPSU.py", line 335, in <module>
    main(sys.argv[1:])
  File "/usr/bin/pyHPSU.py", line 275, in main
    read_can(driver, logger, port, cmd, lg_code,verbose,output_type)
  File "/usr/bin/pyHPSU.py", line 331, in read_can
    hpsu_plugin = module.export(hpsu=n_hpsu, logger=logger, config_file=conf_file)
  File "/usr/lib/python3/dist-packages/HPSU/plugins/mqtt.py", line 77, in __init__
    if self.username:
AttributeError: 'export' object has no attribute 'username'
`

Find here the content of my actual config file:

`# settings for pyHPSUd 
[PYHPSUD]
PYHPSUD_DEVICE=PYCAN
PYHPSUD_LANG=EN
PYHPSUD_PORT=/dev/ttyUSB0

# settings for pyHPSU
[PYHPSU]
PYHPSU_DEVICE = PYCAN
PYHPSU_PORT = /dev/ttyS0
PYHPSU_LANG = DE
OUTPUT_TYPE = MQTT

[CANPI]
timeout=0.05

[MQTT]
BROKER = 192.168.178.10
PORT = 1883
USERNAME = edomi
PASSWORD = 123456
CLIENTNAME = rotex_hpsu
PREFIX = rotex

# interval for each query for a value in seconds
# values not listed here are not queried
[JOBS]
T_HS = 2
T_HS_SET = 25
water_pressure = 4
t_ext = 6
t_dhw = 25
t_dhw_set = 60
t_return = 4
flow_rate = 2
t_hc = 4
t_hc_set = 2`

Is there another config or adjustment I have to do?

Spanni26, 2020-03-29:
Changes mqtt.py a bit...

No Username is never emtpy. If the config can't be read, username is
"None" and a messages is printed.

Please pull the lastest changes.

jockele79, 2020-03-29:
thanks a lot for your patience.

I pulled the latest changes, failure "username not set" appears, but username is definitely set inside config file.
Do I need to insert it in another file?

Spanni26, 2020-03-29:
You run install.sh after download?

jockele79, 2020-03-29:
yes, output changed a bit:

`pi@raspberrypi:/pyHPSU $ pyHPSU.py -c t_hs -o MQTT
Username not set!!!!!
Traceback (most recent call last):
  File "/usr/bin/pyHPSU.py", line 335, in <module>
    main(sys.argv[1:])
  File "/usr/bin/pyHPSU.py", line 275, in main
    read_can(driver, logger, port, cmd, lg_code,verbose,output_type)
  File "/usr/bin/pyHPSU.py", line 332, in read_can
    hpsu_plugin.pushValues(vars=arrResponse)
  File "/usr/lib/python3/dist-packages/HPSU/plugins/mqtt.py", line 92, in pushValues
    self.client.connect(self.brokerhost, port=self.brokerport)
  File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 839, in connect
    return self.reconnect()
  File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 962, in reconnect
    sock = socket.create_connection((self._host, self._port), source_address=(self._bind_address, 0))
  File "/usr/lib/python3.7/socket.py", line 727, in create_connection
    raise err
  File "/usr/lib/python3.7/socket.py", line 716, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
`

Spanni26, 2020-03-29:
Can I have a SSH access to that machine?

Please write me private dspannbauer at web.de

pdcemulator, 2020-03-29:
run
sudo pyHPSU.py -c t_hs ...

jockele79, 2020-03-29:
:-(, same result

`pi@raspberrypi:/pyHPSU $ sudo pyHPSU.py -c th_hs -o MQTT
Username not set!!!!!
Traceback (most recent call last):
  File "/usr/bin/pyHPSU.py", line 335, in <module>
    main(sys.argv[1:])
  File "/usr/bin/pyHPSU.py", line 275, in main
    read_can(driver, logger, port, cmd, lg_code,verbose,output_type)
  File "/usr/bin/pyHPSU.py", line 332, in read_can
    hpsu_plugin.pushValues(vars=arrResponse)
  File "/usr/lib/python3/dist-packages/HPSU/plugins/mqtt.py", line 92, in pushValues
    self.client.connect(self.brokerhost, port=self.brokerport)
  File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 839, in connect
    return self.reconnect()
  File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 962, in reconnect
    sock = socket.create_connection((self._host, self._port), source_address=(self._bind_address, 0))
  File "/usr/lib/python3.7/socket.py", line 727, in create_connection
    raise err
  File "/usr/lib/python3.7/socket.py", line 716, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
`

Spanni26, 2020-03-29:
Connection now is "refused" cause USERNAME is now "None".

jockele79, 2020-03-29:
this refers to the "Username" specified in MQTT-section in pyHPSU.conf?

Spanni26, 2020-03-29:
If no username is found by pyHPSU the Username is set to "None".

pdcemulator, 2020-03-29:
Did you setup config file in git clone folder or in /etc ?

try, here is the right place
 cat /etc/pyHPSU/pyhpsu.conf

Spanni26, 2020-03-29:
That was the problem. He edited the file in the source directory, noch
in /etc/pyHPSU.

I showed him that per teamviewer.

jockele79, 2020-03-29:
problem was sitting in front of the PC, thanks a lot to you

pdcemulator, 2020-03-31:
So clone this issue as it is resolved?

Spanni26, 2020-03-31:
Jup :)

closed 2020-10-02 by Spanni26