Spanni26/pyHPSU #25: Getting lots of "msg not sync" messages

adieball, 2019-08-21:
Hi

running pyHPSU in background, I get a low of the following error messages:

> pi@can:~ $ pyHPSU.py -a &
> [1] 651
> pi@can:~ $ error - SEND:[49, 0, 250, 9, 32, 0, 0]
> error - RECV:bytearray(b' \n\x0c\x00\xf0\x00\x00')
> warning - msg not sync, retry: 1
> error - SEND:[49, 0, 250, 193, 1]
> error - RECV:bytearray(b' \n\x0c\x00\xf0\x00\x00')
> warning - msg not sync, retry: 1
> error - SEND:[49, 0, 250, 6, 108, 0, 0]
> error - RECV:bytearray(b'1\x00\xfa\xc0\xf6\x00\x00')
> warning - msg not sync, retry: 1
> error - SEND:[49, 0, 250, 6, 108, 0, 0]
> error - RECV:bytearray(b'1\x00\xfa\xc0\xf7\x00\x00')
> warning - msg not sync, retry: 2
> error - SEND:[49, 0, 250, 6, 108, 0, 0]
> error - RECV:bytearray(b'1\x00\xfa\xc0\xf8\x00\x00')
> warning - msg not sync, retry: 3
> error - SEND:[49, 0, 250, 6, 108, 0, 0]
> error - RECV:bytearray(b'1\x00\xfa\xc0\xf9\x00\x00')
> warning - msg not sync, retry: 4
> error - SEND:[49, 0, 250, 6, 108, 0, 0]
> error - RECV:bytearray(b'1\x00\xfa\xc0\xfa\x00\x00')
> warning - msg not sync, retry: 5
> error - SEND:[49, 0, 250, 6, 108, 0, 0]
> error - RECV:bytearray(b'1\x00\xfa\xc0\xfb\x00\x00')
> warning - msg not sync, retry: 6
> error - SEND:[49, 0, 250, 6, 108, 0, 0]
> error - RECV:bytearray(b'"\n\xfa\xc0\xf6\x00\x00')
> warning - msg not sync, retry: 7
> error - SEND:[49, 0, 250, 6, 108, 0, 0]
> error - RECV:bytearray(b'"\n\xfa\xc0\xf7\x00\x00')
> warning - msg not sync, retry: 8
> error - SEND:[49, 0, 250, 6, 108, 0, 0]
> error - RECV:bytearray(b'"\n\xfa\xc0\xf8\x00\x00')
> warning - msg not sync, retry: 9
> error - SEND:[49, 0, 250, 6, 108, 0, 0]
> error - RECV:bytearray(b'"\n\xfa\xc0\xf9\x00\x00')
> warning - msg not sync, retry: 10
> error - SEND:[49, 0, 250, 6, 108, 0, 0]
> error - RECV:bytearray(b'"\n\xfa\xc0\xfa\x00\x00')
> warning - msg not sync, retry: 11
> error - SEND:[49, 0, 250, 6, 108, 0, 0]
> error - RECV:bytearray(b'"\n\xfa\xc0\xfb\x00d')
> warning - msg not sync, retry: 12
> error - SEND:[49, 0, 250, 192, 255]
> error - RECV:bytearray(b'1\x00\xfa\xc0\xf6\x00\x00')
> warning - msg not sync, retry: 1
> error - SEND:[49, 0, 250, 192, 249]
> error - RECV:bytearray(b'1\x00\xfa\xc0\xf7\x00\x00')
> warning - msg not sync, retry: 1
> error - SEND:[49, 0, 250, 192, 249]

Spanni26, 2019-08-22:
What adapter are you useing?

adieball, 2019-08-22:
Mqtt

Spanni26, 2019-08-22:
No :) How are you connected to the HPSU :)

adieball, 2019-08-22:
Ups :-)
PiCAN

Spanni26, 2019-08-22:
Hmmm. not bad, not terrible :)
This messages comes up if a lot off values are queried at once. Sometimes the messages get out of the sended order. But I haven't figuered it out deeper.

adieball, 2019-08-23:
ok, thanks

closed 2019-08-23 by adieball