Spanni26/pyHPSU #26: Explanations for values of mode and mode_01

adieball, 2019-08-23:
Hi

is there anywhere an explanation of the values of mode and mode_01?
The xls file with the list pof values is not helpful :-)

svde, 2019-08-23:
From my notes in one of my scripts:

# mode_01;31 00 FA 01 12 00 00;190;i;1;w
# 0100 256 = standby
# 0300 768 = heating
# 0400 1024 = setback
# 0500 1280 = summer
# 0b00 2816 = automatic1
# 0c00 3072 = automatic2
# 1100 4352 = cooling
#mode_01=`/home/stefan/can_progs/can_scan can0 680 180.0112 | grep
"^value" | cut -d " " -f6 | cut -d ")" -f1`

# mode;31 00 FA C0 F6;190;d;1;r
# 0000 = standby, 0001 = heating, 0003 = defrost, 0004 = DHW
#mode=`/home/stefan/can_progs/can_scan can0 680 180.c0f6 | grep
"^value" | cut -d " " -f6 | cut -d ")" -f1`

with kind regards, / met vriendelijke groet,

Stefan van der Eijk

adieball, 2019-08-24:
ok, understood, mode is the general mode and mode_01 the current operating mode.
Mine shows mode = 0 and mode_01 = 5
it's currently set to "cooling"
how do I interpret the numbers:
---cut---
# 0100 256 = standby
# 0300 768 = heating
# 0400 1024 = setback
# 0500 1280 = summer
# 0b00 2816 = automatic1
# 0c00 3072 = automatic2
# 1100 4352 = cooling

svde, 2019-08-24:
Right now my system is in summer mode:

If I retrieve the value using pyHPSU, this is the output:

$ pyHPSU.py -c mode_01
[{'name': 'mode_01', 'resp': '1280', 'timestamp': 1566641041.721531}]

With can_scan this is the output:
$ /home/stefan/can_progs/can_scan can0 680 180.0112 
elster-kromschroeder can-bus address scanner and test utility
copyright (c) 2014 Jürg Müller, CH-5524

value: 0500  (PROGRAMMSCHALTER  Warmwasser)

The pyHPSU output is decimal, can_scan is hexadecimal.