Spanni26/pyHPSU #29: Running pyHPSU

jockele79, 2020-01-02:
Hi,

I'm owner of a HPSU 508 and would like to use pyHPSU to visualize the different parameters in Edomi, unfortunately I have different problems:

pyHPSU is running as a Docker on my Timberwolf Server, the Elm327 interface is connected.
When trying to set the driver:

 "pyHPSU.py -d ELM327 -p var/husky/devices/FTDI_FT232R_USB_UART_A52VL4UF"

 I get back a lot of errors and warnings like:

 "warning - sending cmd 31 00 FA C1 00 (rc:CAN ERROR)
warning - retry 1 command t_r1
warning - sending cmd 31 00 FA C1 00 (rc:CAN ERROR)
warning - retry 2 command t_r1
warning - sending cmd 31 00 FA C1 00 (rc:CAN ERROR)
warning - retry 3 command t_r1
warning - sending cmd 31 00 FA C1 00 (rc:CAN ERROR)
warning - retry 4 command t_r1
error - command t_r1 failed"

Can anyone help me ? 

Thanks a lot

jockele79, 2020-01-24:
Hello,

I'm still trying to install the system properly, unfortunately I have different problems to get pyHPSU running.

Is there a forum or any possibility to get some support ???

Thanks

Spanni26, 2020-01-24:
Only a facebook group....

jockele79, 2020-01-25:
can i enter into this group, is it still active?

svde, 2020-01-25:
The group is called "HPSU Monitoring".

Spanni26, 2020-10-02:
@jockele79 if you don't specify a command, all known commands are querried. Possibly to much for the ELM and the can bus.
Specify some commands with e.g. "-c t_hs" and it will work.
Sorry for the late reply, had this with another user just a few days ago.

closed 2020-10-02 by Spanni26