Spanni26/pyHPSU #32: get value for state of backup heater

pdcemulator, 2020-11-24:
Hi,

I am using pyHPSU for quite a long time, everything is working fine.
But I found that it does not report when the backup heater is running. Based on the docs it should be the ehs value. But its reporting 0, even when the backup heater is active. And the value for qboh is increasing.

Is someone getting data for ehs?
Regarding to #24, maybe ```31 00 FA C0 F9``` is the wrong CAN command?

Spanni26, 2020-11-25:
Hmm,

any Idea how I can test it? At the moment its to warm for the backup heater.

ahermann86, 2020-11-25:
Hello,

the problem is, that you get more than one responses with this request.
You have to parse the buffer complete after a request and split the resulting string with \r.

I dont have time to make this in this phyton script. I made this with perl in my FHEM Module 70_HPSU.pm and it look like this:

`      
..... 
        while($msg =~ m/\r/)
        {
          my $msgSplit = "";

          ($msgSplit, $msg) = split("\r", $msg, 2);
          $msgSplit =~ s/ +$//; #delete whitespace

          my ($name, $nicename, $out) = HPSU_CAN_ParseMsg($hash, $msgSplit);

          if ($name and $out)
          {
            readingsSingleUpdate($hash, "HPSU.$nicename", $out, 1);
            $hash->{jcmd}->{$name}->{FHEMLastResponse} = gettimeofday();
            $hash->{jcmd}->{$name}->{AktVal} = $out; #for "verify"...
            $hash->{helper}{CANRequestPending} = 0;
          }
          $hash->{helper}{CANRequestPending} = 0 if ($hash->{helper}{CANHeaderID} eq "680");  #NoData
        }
`

Maybe someone can fix this with Python in this project...

Greetings 
Axel

Spanni26, 2020-11-25:
Hm,

I get only one message:
   can0  190   [5]  31 00 FA C0 F9
   can0  180   [7]  32 10 FA C0 F9 00 00

But my backup heater is not running. Its to warm here.

pdcemulator, 2020-11-30:
> At the moment its to warm for the backup heater.

For heating, yes. But when the warm water temperature is near the lower threshold and the bathtub is filled... in short the water is tot cold, then the backup heater is starting.
And yes, my goal is to minimize the usage of the heater :-)

ahermann86, 2020-11-30:
Hello,

if you send request: 31 00 FA C0 F9<\r>
youre response will be:
22 0A FA C0 FB 00 64 <\r>
32 10 FA C0 F9 17 70 <\r>
<\r>

17 70 -> 6000 W

This Python script has not implemented the responses who are not requested like "22 0A FA C0 FB 00 64" in this example.
Someone have to fix this in the script...

Hardware:
[image]

My FHEM Module
[image]

Greetings 
Axel

hackepeterOli, 2020-11-30:
Here was my rotex Control unit faulty, pyHPSU ehs works:

Spanni26, 2021-02-11:
Ok, its cold here and the HPSU is using the backup heater.

So I checked it:

root@rotex:~# pyHPSU.py -c ehs -o json
[{'name': 'ehs', 'resp': '9000', 'timestamp': 1613062272.084939}]
root@rotex:~# pyHPSU.py -c ehs -o json
[{'timestamp': 1613062285.739523, 'resp': '0', 'name': 'ehs'}]
root@rotex:~# pyHPSU.py -c ehs -o json
[{'timestamp': 1613062290.793521, 'name': 'ehs', 'resp': '0'}]
root@rotex:~# pyHPSU.py -c ehs -o json
[{'name': 'ehs', 'timestamp': 1613063267.158688, 'resp': '0'}]
root@rotex:~# pyHPSU.py -c ehs -o json
[{'resp': '3000', 'timestamp': 1613064168.072003, 'name': 'ehs'}]

As we can see, at the first shot it runs with 9KW, then it was switched of. On the last shot it was used with 3kW. 
So I think it works as expected.

ckohrt, 2021-03-08:
It works also for me. Same results as above from Daniel.
e.g.
pyHPSU.py -c ehs -d PYCAN -p /dev/ttyS0 -o CSV
1615202591.456367,ehs,0