Spanni26/pyHPSU #33: other options not considered when -a is present

segaura, 2021-02-02:
Hi all.
In automatic mode configuration file has priority over command line options

If my `pyhpsu.conf` has

```
[PYHPSU]
OUTPUT_TYPE = JSON
```

and I specify as command line options `-a -o mqtt`, result is that `-a` works but output is forced in JSON format and I consider this wrong because if I wanted JSON output I would have not set option `-o mqtt`

As far as I can tell this can easily be implemented adding a condition when reading the configuration file checking if options are already present, for instance `output_type` option can be managed in row 172 of `pyHPSU.py`

```
172:        if output_type=="" and config.has_option('PYHPSU','OUTPUT_TYPE'):
173:            output_type=config['PYHPSU']['OUTPUT_TYPE']
```

that in diff world is

```
172c172
<         if config.has_option('PYHPSU','OUTPUT_TYPE'):
---
>         if output_type=="" and config.has_option('PYHPSU','OUTPUT_TYPE'):
```

Same applies to other options, at least `driver`, `port `and `language`.

Spanni26, 2021-02-09:
I'm totally open minded about that. As I always said: pyHPSU isn't "my" project, it is openSource. 
Bit I don't know if it makes sense to provide options in the config file AND the commandline. Od do I understand something wrong?

segaura, 2021-02-09:
Thank you, in my case it does, normally I use default JSON output for single "interactive" commands like -c t_dhw but I want the automatic polling to output to my mqtt queue.

In any case if I specify an option to a command (e.g. -o mqtt) I like to know if this has been considered or not by the command.

If you want me to send the pull request, the commit is this one

https://github.com/segaura/pyHPSU/commit/cd757d4a0338877fbfd444df9af5139d365684dc

and it includes

- _automatic mode: cli options override working_, the one we are talking about
- _help: label lookup through name not command_, a command translation lookup fix regarding the fact that the lookup code supposed that name=command but for some it was not
- _.gitingnore: vscode launch config added_, to avoid committing my vscode launch.json
- _command dictionary EN: scrred -> screed :-)_, cosmetic correction

...I have other [7 commits to propose ](https://github.com/segaura/pyHPSU/commits/master)so I hope you like this one

PR #36 (2021-02-11): Firstpackage
Closes #33

Hallo.
This is the first package of my proposal, it includes only small fixes and small improvements, intended to be easy to review.

**automatic mode**
cli options override working:  if your .conf file says you want JSON output you can write -a -o MQTT and override it
_this solves [issue 33](https://github.com/Spanni26/pyHPSU/issues/33)_

**help**
label lookup through name not command:  lookup is supposed to match label fields between commands_hpsu.json and commands_hpsu_XX.csv but code do not do this, it matches command field from the first and label field from the s [... cut]
closed 2021-02-11 by Spanni26