Spanni26/pyHPSU #40: t_room not working

segaura, 2021-02-17:
Hallo.
I saw in [issue#18](https://github.com/Spanni26/pyHPSU/issues/18) that a commant `t_room` has been added to pyHPSU

```
		"t_room" :{
			"name" : "t_room",
			"command" : "81 00 FA 00 11 00 00",
			"id" : "190",
			"divisor" : "10",
			"writable" : "false",
			"unit" : "deg",
			"type" : "float"
		}
```

I notice that this is giving me error all the times.
Reading [issue#18 comments](https://github.com/Spanni26/pyHPSU/issues/18#issuecomment-492200440), my understanding is that the correct command is

`31 00 FA 00 11 00 00`

I don't have a Rocon U1 so I'm not sure reading this makes sense, anyway with this sequence I get this reading

```
2021-02-17 22:47:59,609 - DEBUG - CanPI, got:  32 10 FA 00 11 80 00
[{'name': 't_room', 'resp': '-3276.80', 'timestamp': 1613598479.611429}]
```

which is better than 45 retries and an error.

Is it correct to change 81 to 31?

Spanni26, 2021-02-18:
Hmmm, could be a typo. I don't know, how I've got this command.....

segaura, 2021-02-18:
It seems you've been given some log [here](https://github.com/Spanni26/pyHPSU/issues/18#issuecomment-492200440)

arizzi, 2022-11-25:
it seems in the testing branch this has been changed to "31" instead of "81" but 31 is not correct, should rather be "61"
The following works:

     "t_room" :{
                        "name" : "t_room",
                        "command" : "61 00 FA 00 11 00 00",
                        "id" : "190",
                        "divisor" : "10",
                        "writable" : "false",
                        "unit" : "deg",
                        "type" : "float"
                },

PR #44 (2021-04-06): mqttdaemon mode
#39 
closes #40 

Hallo.
This pull request adds mqttdaemon mode.
It does this on top of previous proposed improvements, like the log revision.

I'm not particularly happy about three things that I hope can be improved by collaboration:
- modularization, because mqttdaemon has been added in pyHPSU.py trying to keep where possible the previous code structure
- reliability, because mqttdaemon and automatic mode both start threads and I have not checked new and existing code for thread safeness when they work together
- testing, because I do not use mysql, hpsud, elm, backup, resto [... cut]