Spanni26/pyHPSU #41: command list cleanup

segaura, 2021-02-20:
Hallo.
While working on creating a DBC database with the HPSU commands, I've found some maybe quirks in `commands_hpsu.json`.

**Easiest to fix** are these on "screed section":

- these two commands are the same
 `t_screed_day5 61 00 FA 0B BD 00 00`
 `t_screed_day7 61 00 FA 0B BD 00 00`
 - and same for these two
 `t_screed_day21 61 00 FA 0B CD 00 00`
 `t_screed_day22 61 00 FA 0B CD 00 00`

but looking at the progression of the values and also observing that `0x0BBD = 3005` and `0x0BCD = 3021` the correct ones should be

 `t_screed_day7 61 00 FA 0B BE 00 00`
 `t_screed_day22 61 00 FA 0B CE 00 00`

**Second group** are these that are almost the same

`t_flow_cooling 31 00 FA 03 DD 00 00`
`t_flow_cool 61 00 FA 03 DD 00 00`

the first receives a `180#32 10 FA 03 DD 00 BE` as answer and the second receives a `300#32 10 FA 03 DD 00 BE`: is it correct to have them both listed?
I am considering of dropping the second one.

**Third group** is

`quiet_mode 31 00 FA 06 96 00 00`
`aux_fct 31 00 FA 06 96 00 00`

these are also listed as the same command, but after some dump and click I think that this is the correct value

`aux_fct 31 00 FA 06 71 00 00`

**Last group** is this, all three command are the same

`outside_conf 31 00 FA 09 61 00 00`
`storage_conf 31 00 FA 09 61 00 00`
`pres_conf 31 00 FA 09 61 00 00`

and this is correct, they are the same command, but the problem is with the values because they work on one or two bits each and not on one or two bytes like pyHPSU think.

Naming `o` the bit for `outside_conf`, `s` the bits for `storage_conf` and `p` the bit for `pres_conf` the last two bytes (data section 6th and 7th byte) are like this

`0p10100000000sso`

so, for instance

`6804 = 0110100000000100`

corresponds to

- `outside_conf off (0)`
- `storage_conf sensor (4)`
- `pres_conf on (0x6800 = 26624)`

and

`2803 = 0010100000000011`

corresponds to

- `outside_conf on (1)`
- `storage_conf thermostat (2)`
- `pres_conf off (0x2800 = 10240)`

issuing those three commands correctly requires reading the current value, altering the correct bits and sending it back as new value.
This is possible but I have no evidence that this ic currently managed correctly by pyHPSU.
Obviously these three are "less important" commands to control HPSU via pyHPSU so they will probably remain like they are for a while.
They are rather important for backup/restore feature, though, so they deserve some attention, in my opinion.

I have "fixed" first three group in my current development branch, if no objections emerge I can include that fixes in my next PR.

svde, 2021-02-21:
> t_screed_day7 61 00 FA 0B BE 00 00
>

Which is then the same as:
t_screed_day6 61 00 FA 0B BE 00 00

So maybe this is better?

t_screed_day7 61 00 FA 0B BF 00 00

with kind regards, / met vriendelijke groet,

Stefan van der Eijk

segaura, 2021-02-21:
Sure, my typo, you are right, thank you.

Do someone knows if some pyHPSU code deals with single bits (e.g. bit masks) like needed for the `*_conf` commands?

Spanni26, 2021-02-22:
I don't think so.
pyHPSU is not very intelligent. I spits commands to tha can bus and
waits for (more ore less) fixed answers.
I always wan't pyHPSU to build the commands by pieces (like discribed by
the elster telegram http://juerg5524.ch/data/Telegramm-Aufbau.txt ), but
I don't find the time to do it so.