Spanni26/pyHPSU #42: Help needed (can0)

KalaNagHTD, 2021-03-06:
I use pyHPSU on a Pi with Canpi 2.
Whenever I try to retrieve CAN data I get the following message

[image]

With "ifconfig" the can0 doesn't show up

With "dmesg" I find the following lines:
[image]
[image]

Any advice is highly appreciated!

Karl

Spanni26, 2021-03-06:
Am 06.03.2021 um 09:14 schrieb KalaNagHTD:
>
> I use pyHPSU on a Pi with Canpi 2.
> Whenever I try to retrieve CAN data I get the following message
>
> error_message
> <https://user-images.githubusercontent.com/77481230/110200007-c78d9300-7e5b-11eb-9954-f43a79fa8a57.jpg>
>
> With "ifconfig" the can0 doesn't show up
>
> With "dmesg" I find the following lines:
> dmesg1
> <https://user-images.githubusercontent.com/77481230/110200056-28b56680-7e5c-11eb-9695-32a981cafced.jpg>
> dmesg2
> <https://user-images.githubusercontent.com/77481230/110200060-2c48ed80-7e5c-11eb-9f2d-eb95ae007d26.jpg>
>
> Any advice is highly appreciated!
>

Hi Karl,

which distribution are you using? You have to set up a con device first.

Regards

Daniel

KalaNagHTD, 2021-03-06:
Hi Daniel

sorry, with respect to my rookie level I need to have your question and advice a little mor detailed. Could you be so kind?

Karl

martinbischofff, 2021-03-06:
Hi Karl.

Are you talking about PiCAN2? In your post you mentioned Canpi 2 which I can't find.

If you're actually using PiCAN2, did you follow the steps in the software installation manual?
[https://www.skptechnology.co.uk/pican2-software-installation/](url)

Regards
Martin

KalaNagHTD, 2021-03-06:
Hi Martin.

Sorry for the missprint: I use PICAN2! I set it up as described in the manual mentioned above. But I couldn't get the tools working.

The distribution I use is Raspbian 10 (buster)

Regards
Karl

KalaNagHTD, 2021-03-06:
I initialiced can0 again and found the following lines with ifconfig now:

[image]

Nevertheless the error messages remain the same.

KalaNagHTD, 2021-03-06:
...and after a reboot the can0 lines in the ifconfig have disappeared again :-(

KalaNagHTD, 2021-03-06:
Success! After a power reset of my raspi and again initializing the can0 it seems to work. I learned what I should have known: A soft reboot is not enough...

Thanks to both of  you for your support! Issue can be closed!

Karl

closed 2021-03-06 by KalaNagHTD

KalaNagHTD, 2021-03-07:
Wichtiger Nachtrag:
Bei mir fehlten die in Python sehr wichtigen Einrückungen der letzten drei Zeilen im folgenden Code:

auto can0
iface can0 inet manual
     pre-up /sbin/ip link set $IFACE type can bitrate 20000 triple-sampling on
     up /sbin/ifconfig $IFACE up
     down /sbin/ifconfig $IFACE down