Spanni26/pyHPSU #43: one_hot_water not implemented?

ckohrt, 2021-03-06:
Hi! Thanks for your great work!

I tried to srite a value to one_hot_water but received the answer 
root> pyHPSU.py -c one_hot_water:1 -d PYCAN -p /dev/ttyS0
error - type "value" not implemented since yet

Is it planned to be implemented?

Cheers
Christian

ckohrt, 2021-03-06:
Ok, I may have solved it by myself. I have changed the part for one_hot_water in the commands_hpsu.json from:
"one_hot_water" : { 
			"name" : "one_hot_water", 
			"command" : "31 00 FA 01 44 00 00", 
			"id" : "190", 
			"divisor" : "1", 
			"writable" : "true", 
			"unit" : "longint",
			"type" : "value",
			"value_code" : {
				"0" : "off",
				"1" : "on"
			}
to:
"one_hot_water" : { 
			"name" : "one_hot_water", 
			"command" : "31 00 FA 01 44 00 00", 
			"id" : "190", 
			"divisor" : "1", 
			"writable" : "true", 
			"unit" : "longint",
			"type" : "longint",
			"value_code" : {
				"0" : "off",
				"1" : "on"
			}

Can somebody with more experiance than me check it and change it in the code please?

segaura, 2021-03-07:
> Hi! Thanks for your great work!
> 
> I tried to srite a value to one_hot_water but received the answer
> root> pyHPSU.py -c one_hot_water:1 -d PYCAN -p /dev/ttyS0
> error - type "value" not implemented since yet
> 
> Is it planned to be implemented?
> 
> Cheers
> Christian

I think the general request ultimately relates to [this](https://github.com/Spanni26/pyHPSU/issues/41#issuecomment-783149592)  kind of improvements

ckohrt, 2021-03-07:
#45

Spanni26, 2021-03-07:
I modified the commands_hpsu.json and the pyHPSU.py to work with type =
value.

Should work now. But there is a problem when setting one_hot:water to
"0" or "off". I get can bus errors, but I don't know why.

regards

Daniel

ckohrt, 2021-03-07:
Hi Daniel,
I will never ever set one_hot_water to "0", because
1. the pump SHALL do one time hot water generation up to the defined temperature and stops automatically.
2. this is not possible from the Daikin Control GUI
There is just no reason to do so. Maybe the pump just doesn't accept "0".

ckohrt, 2021-03-07:
from my side we can clode this issue

ckohrt, 2021-03-07:
just in case somebody thinks that this is working. I tried it today in my automation system. Result: mode was set to 4 but I have no evidence that it really did hot water.... So I check tomorrow again.

ckohrt, 2021-03-08:
I checked the new implementation - it works! Thanks!

ckohrt, 2021-03-08:
> I think the general request ultimately relates to [this](https://github.com/Spanni26/pyHPSU/issues/41#issuecomment-783149592) kind of improvements

thanks for linking

closed 2021-03-08 by ckohrt