Spanni26/pyHPSU #46: ELM not working

m-reuter, 2021-03-12:
Hi, can anyone help me get started?

I bought the ELM that is linked in the readme, 
"bbfly Forscan Elm327 ELMconfig USB VINT-TT55502 mit MS-CAN für Ford OBD2 Diagnose"
but cannot get it to work (actually I tried with two of these, to make sure it is not broken - bought from different vendors, but otherwise the same ). Both show the same problems. Also, my heat pump is Rotex HPSU compact ultra. I know the folks at FHEM use the same adapter and it works for them. My goal is to run pyHPSU in a docker and push info (both ways) via MQTT to homeassistant. 

I usually get (but not always) when I run for the first time after connecting:
 pyHPSU.py -d ELM327 -p /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 -v 2 -c flow_rate
error - Error sending AT PP 2F ON (rc:AT PP 2F ONOK)

and when I run it a second time:
 pyHPSU.py -d ELM327 -p /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 -v 2 -c flow_rate
warning - sending cmd 31 00 FA 01 DA 00 00 (rc:NO DATA)
warning - retry 1 command flow_rate
(also when trying other commands)

I have a Raspberry Pi 4 B. Is the ELM working for anyone in this constellation? What ELM are you using? 
Or do I need to get a PICAN hat? If so, does the PICAN 2 work on the Raspi 4? 

Thanks for any help.

P.S. I connected to Rotex first from the female OBD 16 pin connector with 2m wire. Then, because I was unsure if wires are twisted pairs in that cable, I now use an ethernet cable with a twisted pair. Either way same result. Also, I run pyHPSU in a docker, but given that the ELM reacts I think that works fine. In the meantime, I will re-build that docker to be double sure that I am running the latest version but I built it just a couple of weeks ago.

KalaNagHTD, 2021-03-13:
I tried the same ELM327 adapter and got the same result. :-(
The command „AT Z“ is answered with „ELM327 v1.5“.
Zanac wrote that Version 1.5 is a China clone where the PP command is not implemented completely.
Can anybody with a working adapter confirm that? What version should a working adapter have?

regards Karl

martinbischofff, 2021-03-13:
> 
> Or do I need to get a PICAN hat? If so, does the PICAN 2 work on the Raspi 4?
> 

I'm using a Raspi 4 with the PICAN 2 duo. Works without any problems.

Regards
Martin

m-reuter, 2021-03-13:
Today I tried the RS485 CAN Hat also without any success, although I am not sure if it can be seen, because either way (conntected to rotex or not) I get this:
pyHPSU.py -c t_hs
error - Not aquired bus
warning - msg not sync, retry: 1
...

I do see the can0 in ifconfig (outside the docker) but I don't have ifconfig in the docker image, to check there. 

Another thing: 

When I connect the Rotex this happens (also occured with the ELM before): 
Rotex goes into "Absenk" (sorry, don't know the english)  mode and I cannot see any room temp (only "-") I have to restart the Rotex again after that. This behavior is a little scary. It happens without sending a command, only attaching the two (hi, lo) CAN cables.

KalaNagHTD, 2021-03-13:
With ifconfig my **Pican2** shows up as follows:

can0: flags=193<UP,RUNNING,NOARP>  mtu 16
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 10  (UNSPEC)
        RX packets 2682935  bytes 18780545 (17.9 MiB)
        RX errors 1  dropped 1279  overruns 0  frame 1
        TX packets 166256  bytes 1163792 (1.1 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

with dmesg I see the following:

[    8.342024] CAN device driver interface
[    8.359017] mcp251x spi0.0 can0: MCP2515 successfully initialized.

Does your **Waveshare RS485 CAN Hat** show similar lines? 
Did you modify the cofig.txt according to the RS485-CAN-HAT-user-manual?(https://www.waveshare.com/w/upload/2/29/RS485-CAN-HAT-user-manuakl-en.pdf))

KalaNagHTD, 2021-03-14:
I forgot to mention:
The error message „ error - Not aquired bus“ shows up, when the CAN board is not initialiced correctly in the config.txt

m-reuter, 2021-03-14:
Thanks Karl, 

I had the config.txt adjusted already: 

# RS485 CAN Hat
# https://www.waveshare.com/w/upload/2/29/RS485-CAN-HAT-user-manuakl-en.pdf
dtparam=spi=on
dtoverlay=mcp2515-can0,oscillator=8000000,interrupt=25,spimaxfrequency=1000000

Ifconfig shows only zeros (which could be ok after a reboot without sending anything). 

can0: flags=193<UP,RUNNING,NOARP>  mtu 16
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 10  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

And I had this (same as in README) in /etc/network/interfaces also:

auto can0
iface can0 inet manual
     pre-up /sbin/ip link set $IFACE type can bitrate 20000 triple-sampling on
     up /sbin/ifconfig $IFACE up
     down /sbin/ifconfig $IFACE down

I'll probably have to try outside of docker, if I start inside docker I get the errors and also the TX RX packages do not change. 
This is my docker command 
docker run --name pyhpsu -v /home/pi/projects/pyHPSU-docker-cmds:/etc/pyHPSU --network host -it  --rm  --entrypoint /bin/bash pyhpsu

and I pass -network host to have can0 available inside docker (maybe that part is not working properly). Maybe I can test this with picocom but not sure how I connect it to the can0 device, it only worked with /dev devices so far. 
thanks.

KalaNagHTD, 2021-03-14:
> I tried the same ELM327 adapter and got the same bad result. :-(
> The command „AT Z“ is answered with „ELM327 v1.5“.
> Zanac wrote that Version 1.5 is a China clone where the PP command is not implemented correctly.
> Can anybody with a working adapter confirm that? What version should a working adapter have?
> Pics of my board showing the microprozessor PIC18F25K80 that emulates the ELM327:
> 
> [image]
> [image]
> 
> regards Karl

With regard to the findings in the project „fhemHPSU“ my Adapter ELM v.1.5 with the PIC18F25K80 should do the job!

m-reuter, 2021-03-16:
So what do people get when they run pyHPSU without anything attached to the CAN interface? Is it the same as
```
exception - Error sending msg
error - Not aquired bus
warning - msg not sync, retry: 1
error - Not aquired bus
...
```

and is there any way to check if my can0 interface works without connecting it? Then I'd know if I need to debug on the raspi side or if it could be the cable connection (at least for the can hat version). For ELM I think the problems is likely on the software/driver/raspi side given that:
- I can contact it (e.g. via picocom ) and interact
- pyHPSU works for KarlaNag when he uses the can0 interface but same cable connection attached to ELM does not work
- and it should work theoretically according to the readme and according to the people at fhemHPSU

Also I modified pyHPSU.py  to work  from a local directory without having to install it, so I can test without docker. Only had to change the hard-coded paths everywhere. 

Any help would be apprechiated. Thanks.

KalaNagHTD, 2021-03-17:
> So what do people get when they run pyHPSU without anything attached to the CAN interface? Is it the same as
> 
> ```
> exception - Error sending msg
> error - Not aquired bus
> warning - msg not sync, retry: 1
> error - Not aquired bus
> ...
> ```
> 

If I rember correctly I got the same message when my Pican2 was not initialiced yet. It had nothing to do with the CAN connection. Did you activate ILS with raspi-config?

KalaNagHTD, 2021-03-17:
In order to clarify that I removed the CAN-connection from my Pican2 and got the same message („Error - Not aquired bus“)

m-reuter, 2021-03-17:
Thanks @KalaNagHTD for testing! In the meantime I obtained a Pican2  hat (my 4th CAN device after two ELMs and the RS485 CAN Hat) and with the Pican2 on Raspi 4b it finally works (both inside and outside docker).  

Given that there was little response here with respect to those other two devices I would advise to remove those buying recommendations from the README until we know what is causing these problems: Could be changes in the code, could be incompatibility with some more recent RasPi (I have 4b) or incompatibility with the compact ultra model of Rotex. We don't know. 

Also it would be great if anyone who has a running setup with either the ELM or the RS485 CAN Hat can post details of his Hardware (exact CAN device, computer, and heatpump model) and Software setup here. 

Happy to try a couple things, will probably keep one of the ELMs around and will return the RS.. hat and 2nd ELM.

evildad, 2021-12-22:
Just wanted to say that the RS485 CAN Hat works fine with my old Pi2 which I'm using.
Added the config to /boot/config.txt and the rest was straight forward.
`# CAN
dtparam=spi=on`
`dtoverlay=mcp2515-can0,oscillator=12000000,interrupt=25,spimaxfrequency=1000000`

KalaNagHTD, 2021-12-22:
Good news for budget sensitive projects! Thank you!

acalatrava, 2022-07-28:
> Just wanted to say that the RS485 CAN Hat works fine with my old Pi2 which I'm using.
> Added the config to /boot/config.txt and the rest was straight forward.
> `# CAN`
> `dtparam=spi=on`
> `dtoverlay=mcp2515-can0,oscillator=12000000,interrupt=25,spimaxfrequency=1000000`

Hi! I want to connect to my HPSU too and I have the RS485 Can Hat. Can you tell me which connections did you use? I mean, I can see that we have the CAN-H and CAN-L on the RS485 Hat but not the GND. I guess I can use any GND from the Pi? Or GND is not needed?
Thanks!

evildad, 2022-08-03:
Hi @acalatrava ,
sorry for my late answer, CAN-H and CAN-L with twisted pair cable is working for me, GND is not needed.

ap20132022, 2023-05-14:
> Hi, can anyone help me get started?
>
> I bought the ELM that is linked in the readme, "bbfly Forscan Elm327 ELMconfig USB VINT-TT55502 mit MS-CAN für Ford OBD2 Diagnose" but cannot get it to work [...]

I have the same issue with my USB PIC18F25K80.
How about you? Do you still use it? I remember that someone has used it with fhem.

m-reuter, 2023-05-14:
Not sure about your ELM. I have rotex compact ultra also and used the pican 2 hat with success in the past. I also created an Addon for Homeassistant: 
https://github.com/m-reuter/ha-addons/tree/master/pyhpsu2mqtt 

However after updating my PI 4 to Debian for a supported Homeassistant install, I cannot reach my pican any longer (I think drivers are missing from the kernel). So far I could not get that fixed, was planing on waiting for support in a newer kernel version, which I think is on the release list. If anyone has a pican 2 hat running with debian, let me know how you did it. Thanks.

Vortecsz, 2023-07-24:
I'm having the same problem @m-reuter. Using RS485 CAN HAT with pi3b+. Exactly the same problem; exception - Error opening bus can0. I don't know what to do

svde, 2023-07-25:
Do you have the right frequency in /boot/config.txt? I have 2 CAN HATs, with different crystals on it. The old one needs:

dtoverlay=mcp2515-can0,oscillator=8000000,interrupt=25,spimaxfrequency=1000000

and the newer one:

dtoverlay=mcp2515-can0,oscillator=12000000,interrupt=25,spimaxfrequency=2000000

Vortecsz, 2023-07-25:
Yes I have the one with the 12000 crystal. And I have the correct configuration. Did you do anything else to make it work?

Vortecsz, 2023-07-27:
@svde

svde, 2023-08-19:
$ cat /etc/network/interfaces.d/can0
auto can0
iface can0 inet manual
     pre-up /sbin/ip link set $IFACE type can bitrate 20000 triple-sampling on
     up /sbin/ifconfig $IFACE up
     down /sbin/ifconfig $IFACE down

m-reuter, 2023-11-25:
Hi, I am running Debian (instead of raspberry pi OS) on my raspberry pi. The problem is that the Debian kernels do not support device tree overlays, and so even if the CAN kernel module is loaded, it cannot find the PiCan 2 hat. 

Does anyone have a CAN hardware (e.g. via USB) that works with Debian (bookworm)?

evildad, 2023-11-26:
Hi @m-reuter , I'm using the following [Adapter](https://www.fysetc.com/en-de/products/fysetc-ucan-board-based-on-stm32f072-usb-to-can-adapter-support-with-canable-candlelight-klipper-firmware) which you can get relatively cheap from Aliexpress (~13€) on my Debian (bookworm) Server. There's also a pretty nice looking 3D printable Case [available](https://www.thingiverse.com/thing:5874919).
The main reason for me was getting rid of the additional Raspberry PI and have everything together containerized on my Home Server.

m-reuter, 2023-12-09:
Thanks @evildad . I also bought that device, and can see the can0 interface directly. I brought it up with 
```
sudo ip link set can0 type can bitrate  20000 
sudo ip link set up can0
 ```
and can see it with `ip a`: 

```
can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP group default qlen 10
    link/can
```

But pyhpsu does not seem to be able to send or receive anything. With errors like: 

```
2023-12-09 13:19:48,349 - ERROR - CanPI t_dhw, msg not sync, timeout
2023-12-09 13:19:50,177 - ERROR - Error sending msg
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/can/interfaces/socketcan/socketcan.py", line 778, in _send_once
    sent = self.socket.send(data)
OSError: [Errno 105] No buffer space available
```

How did you set it up? What bitrate did you use?

evildad, 2023-12-09:
Hey @m-reuter please try it again with the pyHPSU devel branch commit 34d6afba6b85c1085fc88d0454b9c7a3b646bc5c and I have also pinned python-can==3.3.4 in my Dockerfile (FROM python:3.9-alpine).
I had problems with everything newer even with my old Raspberry Pi setup but haven't digged deeper into the Problem.

evildad, 2024-01-12:
Hey @m-reuter , did it work? Or did you get the same errors?

m-reuter, 2024-01-13:
Thanks for asking, I could not get it running yet:
- how do you bring up your can0 ? Any specific bitrate you specify there?
- can you send your docker file? In mine, many dependencies are newer and not compatible with the old commit that you list. 
Thanks

m-reuter, 2024-01-13:
Oh, actually it seems to be working kind-of. I get lots of errors like this, maybe it does not really do any buffering. But it does send data via MQTT.

 Are you using the pyhpsud (deamon)?

```
2024-01-13 16:45:40,415 - ERROR - Error sending msg
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/can/interfaces/socketcan/socketcan.py", line 778, in _send_once
    sent = self.socket.send(data)
OSError: [Errno 105] No buffer space available

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/HPSU/canpi.py", line 88, in sendCommandWithID
    self.bus.send(msg)
  File "/usr/lib/python3.10/site-packages/can/interfaces/socketcan/socketcan.py", line 763, in send
    sent = self._send_once(data, channel)
  File "/usr/lib/python3.10/site-packages/can/interfaces/socketcan/socketcan.py", line 780, in _send_once
    raise can.CanOperationError(
can.exceptions.CanOperationError: Failed to transmit: No buffer space available [Error Code 105]
```

m-reuter, 2026-08-25:
Oh, in the meantime I got it working (usb can) both on Debian bookwork and raspberry pi os (trixie) . Not sure any more why it failed earlier. I just brought up the device and it works, see documentation in my home assistant add-on: https://github.com/m-reuter/ha-addons/tree/main/pyhpsu2mqtt