Spanni26/pyHPSU #47: Autostart after reboot

KalaNagHTD, 2021-03-18:
I‘d like to autostart pyHPSU after reboot of my Raspberry Pi. Can anyone help me with a working procedure?

regards Karl

Spanni26, 2021-03-18:
With wihich paramaters shozuld it be started?

Regards

Daniel

KalaNagHTD, 2021-03-18:
pyHPSU.py -a -o MQTT

Spanni26, 2021-03-18:
Hello,

set "OUTPUT_TYPE = MQTT" in /etc/pyHPSU/pyhpsu.conf, so you don't have 
to set it at the command line.

There is a systemd service for pyHPSU.
To Enable ist:
systemctl enable hpsu.service
To start it:
systemctl start hpsu.service

Thats all.

Regards

Daniel

martinbischofff, 2021-03-18:
I'm using a script in /etc/init.d that calls pyHPSU.py -a.

Don't know whether Daniel's approach is better or whether it's more or less the same.

Regards
Martin

KalaNagHTD, 2021-03-18:
Defining MQTT output in pyhpsu.conf
configparser.MissingSectionHeaderError: File contains no section headers.
file: '/etc/pyHPSU/pyhpsu.conf', line: 1

closed 2021-03-18 by KalaNagHTD

Spanni26, 2021-03-18:
Can you show us your pyhpsu.conf?

Spanni26, 2021-03-18:
/etc/init.d works only for compatibility. Nearly every linux
distribution uses systemd. But systemd can work with initd scripts.

KalaNagHTD, 2021-03-18:
> […](#)

> […](#)

Here is my pyhpsu.conf. I had added the "OUTPUT_TYPE = MQTT" as the first line (removed here):

pi@raspberrypi:/etc/pyHPSU $ cat pyhpsu.conf
# settings for pyHPSUd
[PYHPSUD]
PYHPSUD_DEVICE=PYCAN
PYHPSUD_LANG=EN
PYHPSUD_PORT=/dev/ttyUSB0

# settings for pyHPSU
[PYHPSU]
PYHPSU_DEVICE = PYCAN
PYHPSU_PORT =
PYHPSU_LANG = DE
OUTPUT_TYPE = MQTT

[CANPI]
timeout=0.05

[MQTT]
BROKER = localhost
PORT = 1883
USERNAME = pi
PASSWORD = raspberry
CLIENTNAME = rotex_hpsu
PREFIX = rotex

# interval for each query for a value in seconds
# values not listed here are not queried
[JOBS]
bpv = 5
posmix = 5
flow_rate = 5
water_pressure = 60
t_dhw_set = 60

Spanni26, 2021-03-18:
" I had added the "OUTPUT_TYPE = MQTT" as the first line (removed here):"

Thats wrong. It hast to be under [PYHPSU]
In the conf you posted, it is alread configured.

Am 18.03.2021 um 14:02 schrieb KalaNagHTD:
> I had added the "OUTPUT_TYPE = MQTT" as the first line (removed here):

KalaNagHTD, 2021-03-18:
Sorry, I accidentally managed to close this issue :-(

MQTT doesn‘t work with just the setting in the pyhpsu.conf as shown above. I need to specify MQTT in the command line. Otherwise I get the default JSON output. I have no idea why!

Spanni26, 2021-03-18:
Hmmm, possibly a bug. I will inspect this.
You can modify /etc/systemd/system/hpsu.service to youre needs and you
have to call "systemctl --system daemon-reload"
and "systemctl restart hpsu.service"

Regards

Daniel

KalaNagHTD, 2021-03-18:
Many thanks, Daniel!
I added „-o MQTT“ to the hpsu.service and got it running :-)

regards Karl

segaura, 2021-03-26:
IMHO the power of systemctl is in its ability to restart the process when something bad happened, and also to enforce an order between various services at startup, this is my modified `hpsu.service` file.
Please note the `Restart=on-failure` and the vaious `After` instructions

```
[Unit]
Description=HPSU in auto mode with output to influxDB and MQTT plus MQTT Daemon mode activated
Requires=canable.service
After=canable.service
After=network-online.target

[Service]
Type=simple
ExecStartPre=
#EnvironmentFile=/etc/pyHPSU/pyhpsu.conf
ExecStart=python3 /usr/bin/pyHPSU.py -a -o influxdb -o mqtt --mqtt_daemon --log_level WARNING --log /tmp/pyHPSU.log
ExecReload=
ExecStop=
Restart=on-failure
WorkingDirectory=/root
User=root

[Install]
WantedBy=multi-user.target
```