Spanni26/pyHPSU #49: update commands_hpsu.json

shortcircuit0815, 2021-04-15:
would you please update again the "commands_hpsu.json" with some not yet existing commands?
I updated my own local one with e.g. "comp_aktiv" and "pump_aktiv" from [FHEM project](https://github.com/ahermann86/fhemHPSU/blob/main/commands_hpsu.json) and it works fine.

For me the current status (on/off) of the compressor and the pump is a very useful information and I think it would be a benefit for all others.

Thanks

segaura, 2021-06-12:
They are already present, I've taken them from that or a similar source a few months ago.
Have a look at the `testing` branch, [here](https://github.com/Spanni26/pyHPSU/blob/testing/etc/pyHPSU/commands_hpsu.json)

There they are:

```
		"comp_active" : {
			"name" : "comp_active",
			"command" : "A1 00 61 00 00 00 00",
			"id" : "190",
			"divisor" : "1",
			"writable" : "false",
			"unit" : "",
			"type" : "longint",
			"value_code" : {
				"off" : "0",
				"on" : "256"
			}
		},
		"pump_active" :{
			"name" : "pump_active",
			"command" : "A1 00 FA FD AC 00 00",
			"id" : "190",
			"divisor" : "1",
			"writable" : "false",
			"unit" : "",
			"type" : "longint",
			"value_code" : {
				"off" : "0",
				"on" : "256"
			}
		},
```

dunno when `testing` branch will reach `master`, test it you too to speed up things, maybe

Spanni26, 2021-06-12:
Hi,

if testing is good so far, we can merge it to master.

shortcircuit0815, 2021-06-23:
one more question before merging.
"func_heating" does not work for writing. Furthermore i changes the settings because i got an error when using "value" as type.

		"func_heating" : { 
			"name" : "func_heating", 
			"command" : "A1 00 FA 06 D2 00 00", 
			"id" : "190", 
			"divisor" : "1", 
			"writable" : "true", 
			"unit" : "",
			"type" : "longint"
		},
But it tunred out that this is also not the solution. Now I'm able to set func_heating to 1, 2 and three but it is not possible to deactivate it by choosing 0.
When setting 0 manually at the device then 0 can be read via pyHPSU. If you then again set via PyHPSU to e.g. 1 and then back to 0 it remains at 1.

Any ideas? Does the device block commands with 0?

Spanni26, 2021-06-23:
Hmmm,

I always start a candump and do change the setting at the ROCON. Then I
can see the command the ROCON sends to the HPSU. So I can check which
value is needed.

Regards

Daniel

shortcircuit0815, 2021-09-01:
sorry for late answer.
I double checked by using candump and found out the following:

If you read the value for func_heating the first Byte is A1, if you write it's A2
read func_heating:      can0  190   [7]  A1 00 FA 06 D2 00 00
set func_heating to 1: can0  680   [7]  A2 00 FA 06 D2 00 01
set func_heating to 2: can0  680   [7]  A2 00 FA 06 D2 00 02
...

but if you want to set it to 0 it looks like a read command:
can0  190   [7]  A1 00 FA 06 D2 00 00

The command in ssh is:
pyHPSU.py -c func_heating:0
pyHPSU.py -c func_heating:1
...

Could it be that it is not allowed/intendet to set a value = 0?

segaura, 2022-01-22:
I have not written that code so I am not aware of all the possibilities that needs to be managed, nor I am enough familiar with python operators to propose a solution.

Instead of using a separate script, aren't you able to modify that row #303 to test if your use case works and the other use cases are not broken?

I mean something like

`       if not (setValue = "0") and not setValue:`

If this or something similar works, you can submit a pull request or ask here someone else to do it here.

Spanni26, 2022-02-04:
Perhaps it would be good to define in commands_hpsu.json which values are allowed and which not.