Spanni26/pyHPSU #52: python-can==4.0.0 not working

m-reuter, 2022-02-24:
Hi, for me the latest python-can  4.0.0 is not working. I am getting: 

`ERROR - Error opening bus can0
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/HPSU/canpi.py", line 23, in __init__
    self.bus = can.interface.Bus(channel='can0', bustype='socketcan_native')
  File "/usr/lib/python3.9/site-packages/can/interface.py", line 101, in __new__
    kwargs = load_config(config=kwargs, context=context)
  File "/usr/lib/python3.9/site-packages/can/util.py", line 192, in load_config
    bus_config = _create_bus_config(config)
  File "/usr/lib/python3.9/site-packages/can/util.py", line 211, in _create_bus_config
    raise CanInterfaceNotImplementedError(
can.exceptions.CanInterfaceNotImplementedError: Unknown interface type "socketcan_native"`

It does work with the version before: python-can-3.3.4 

Anyone else experiencing this? It would be great to have a requirements file with the supported versions.

Spanni26, 2022-02-24:
Hi,

4.0.0 was released 6 days ago. I don't think that anyone tried this
version so far.
Feel free to mark this at the readme.md

Regards

Daniel

m-reuter, 2022-02-25:
OK, I fixed it. Basically we were using deprecated socketcan_native which was removed in v. 4 and also extended_id (now is_extended_id ) , see https://python-can.readthedocs.io/en/master/interfaces/socketcan.html 
Should I generate a pull request to master or testing or both?

Spanni26, 2022-02-25:
Hi,

thanks for the work.
One question: ist in compatible backwards with older python-can? Oder
does ist only work with 4.0.0?

m-reuter, 2022-02-25:
Tested it with 4.0.0, 3.3.4 and 3.0.0 and it is working for me. I think these things were deprecated for a while and were finally removed. I will generate pull request to testing, which I hope can be merged into master soon. Or is there anything that works in master but not in testing currently?

Spanni26, 2022-04-07:
Which distribution are do you use?

m-reuter, 2022-04-07:
alpine 3.13 or higher (not sure any more, it is in a docker image). 
https://hub.docker.com/r/mreuter/armv7-pyhpsu2mqtt 
as a HomeAssistant AddOn.

Spanni26, 2022-05-19:
I have to reopen it...
I stumbled on a users machine with installed python-can 4.0.0

`Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/HPSU/canpi.py", line 23, in __init__
    self.bus = can.interface.Bus(channel='can0', bustype='socketcan_native')
AttributeError: module 'can' has no attribute 'interface'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/HPSU/canpi.py", line 23, in __init__
    self.bus = can.interface.Bus(channel='can0', bustype='socketcan_native')
AttributeError: module 'can' has no attribute 'interface'
`

Seems, that 4.0.0 has no "interface" attribute. 
Needs a second look.

m-reuter, 2022-05-19:
Are you using the testing branch? Based on the error message I think not.  In my fix in #54 I had changed the bus type to socetcan (socketcan_native has been removed in 4.0 after a long depreciation period). 
https://github.com/Spanni26/pyHPSU/blob/de88ca7dbd42cb5b0d7d95a8706fa99c953c4169/HPSU/canpi.py#L23 

About "interface" it should work (at least it worked for me on 4.0 and also one of the examples in the above socket can link uses can.interface.bus (scroll down to broadcast manager). I am not familiar with socket can, so not sure what will happen if you simply remove the "interface". But first test if this is not a different problem (Bustype).

Spanni26, 2022-05-19:
Tried it with a python3-can installed via pip....
A quick look at the manpage also says nothing about "interface"...

Regards

Daniel

Spanni26, 2022-05-19:
Sorry, thats wrong....
Even the example on the python-can doc shows the "interface" attribute. Wtf.....

m-reuter, 2022-05-19:
Yes, but your error message
` line 23, in init self.bus = can.interface.Bus(channel='can0', bustype='socketcan_native')` 
indicates that you are not using the testing branch, but maybe master? There should be no socketcan_native in our code any longer.

svde, 2024-01-08:
The python3-can package was updated on my raspberry pi, had to change

From:
             self.bus = can.interface.Bus(channel='can0', bustype='socketcan_native')

To:
            self.bus = can.Bus(channel='can0', interface='socketcan')